nordicsemi/nRF-Sniffer-for-802.15.4 #8: Wireshark Crash when Capture starts

Clem-, 2018-12-08:
I've flashed my nRF52840dk board with JFlashLite on macOS Mojave.
After copying `nrf802154_sniffer.py` in extcap directory of Wireshark 2.6.4, and modifying its first line with `#!/usr/local/bin/python` the "nRF 802.15.4 sniffer" interface appears in Wireshark.
Under interface options, I've selected `Device=/dev/cu.usbmodem` and let `Channel=11`.

Unfortunately, Wireshark crashes when starting the capture.

stig-bjorlykke, 2018-12-09:
Do you get a stack dump from the Wireshark crash? I believe this is a Wireshark issue and you should create a bug report in the [Wireshark Bug Database](https://bugs.wireshark.org/bugzilla/).

Do you know why you need to use ```/usr/local/bin/python```? The version shipped with macOS Mojave should work.

IanAber, 2019-01-18:
I just downloaded the development release of Wireshark 2.9.0 and replace my 2.6.6 copy. The nRF52840 dongle is now sniffing Zigbee packets. I guess the problem is with Wireshark somewhere.

stig-bjorlykke, 2019-02-25:
Wireshark is close to the version 3.0 release. Are any of you able to test the [latest RC](https://www.wireshark.org/download/osx/) version to verify that the crash issue is fixed?

e-rk, 2019-08-06:
Duplicate of  #11.

closed 2019-08-06 by e-rk