nordicsemi/nRF-Sniffer-for-802.15.4 #9: nRF52840 Dongle support

istirbu, 2018-12-14:
Hello,

Is it possible to get a hex for nRF52840 Dongle?

Thanks!

e-rk, 2018-12-14:
Hi,
Hex for nRF52840 dongle will hopefully be coming after this weekend.

Cheers.

istirbu, 2018-12-14:
@e-rk Thanks a lot!

e-rk, 2018-12-18:
Hi,
Here (https://github.com/NordicPlayground/nRF-802.15.4-sniffer/pull/10) is a current pull request with dongle support.
Feedback is very welcome.

istirbu, 2018-12-18:
Hi @e-rk, thanks for the update. But I'm missing something, I can't flash program the dongle with nRF Connect - Programmer app, I have only the reset option enabled. Can you help with any information?

e-rk, 2018-12-19:
Hey,
Have you selected `nrf802154_sniffer_dongle.hex` file? It looks like the programmer doesn't allow flashing the device when a wrong file is chosen, because it would overwrite the MBR section. Changes from the PR I linked earlier have been merged today so please pull the newest commit if you haven't already.
Hope it helps.

istirbu, 2018-12-19:
I was able to flash the dongle firmware, and verified that the start/end addresses and size match after flashing. Though, unplugging and plugging back did not make the dongle visible, still had to press the reset button. After the reset, the led is blinking red. Not sure if it matters. 

I've upgraded to wireshark 2.6.3 (Win 10) and copied the py/bat files to extcap. The plugin was detected properly, but after the capture starts, no data is detected (I have a couple of zigbee devices around). 

I've tried to run the python script from a Linux VM to test the output, but I get the same error as in Wireshark:
```
python nrf802154_sniffer.py --dev=/dev/ttyACM0 --channel=11 --extcap-interface=nrf802154 --fifo='/tmp/nrf' --capture
2018-12-19 16:18:51,585 [ERROR] Nrf802154Sniffer (/dev/ttyACM0) channel 11 did not reply properly to setup commands. Is it flashed properly? Recieved: 
```
And after the colon there is no data.

e-rk, 2018-12-19:
The reset button should not be clicked after flashing because then the device enters DFU mode and sniffer application is not running.
When you plug the device does anything appear in Windows device manager?

istirbu, 2018-12-19:
The dongle does not appear in the Ports list when plugged in ([first picture](https://imgur.com/a/wAIlLrH)). But after pressing reset, in DFU mode it's listed properly ([second picture](https://imgur.com/a/wAIlLrH)).

e-rk, 2018-12-20:
I have checked on Windows 10 as well and indeed the device does not communicate with the system correctly due to driver issue. Currently I don't have an option to check how it works on older Windows operating systems, but I can confirm that Ubuntu 18.04 running natively is able to talk with the device.
I have also tested other USB applications on Win10 and they too have the same problem with the USB driver, which indicates a broader issue.
Unfortunately there is not much I can do at this point until beginning of January.

Sorry for that and have a great holidays.

e-rk, 2019-01-07:
I think the sniffer should be ready to run with Wireshark now.
The COM9 port should be responsible for communication with the sniffer application.
The second interface with the warning sign should be the DFU trigger interface, which enables the dongle to enter DFU mode after receiving a signal over USB, without having to press reset button. I'm not sure why the automatic update doesn't select the correct driver for this one. You can either ignore it or you can try finding and manually selecting Nordic DFU trigger driver in device properties. I think the driver should come with nRF Connect for desktop.

istirbu, 2019-01-15:
hi @e-rk , the dongle works fine and I can see the traffic! Thank you!

closed 2019-01-15 by istirbu