nordicsemi/nRF-Sniffer-for-802.15.4 #11: Using Wireshark on Mac errors with nRF52840_sniffer.py script

IanAber, 2019-01-23:
Using Wireshark 2.9.0 on Mohave I am able to start a capture but I see these errors thrown from the Python script...

usage: nrf802154_sniffer.py [-h] [--extcap-interfaces]
                            [--extcap-interface EXTCAP_INTERFACE]
                            [--extcap-dlts] [--extcap-config]
                            [--extcap-reload-option EXTCAP_RELOAD_OPTION]
                            [--capture] [--fifo FIFO]
                            [--extcap-capture-filter EXTCAP_CAPTURE_FILTER]
                            [--extcap-control-in EXTCAP_CONTROL_IN]
                            [--extcap-control-out EXTCAP_CONTROL_OUT]
                            [--channel CHANNEL] [--dev DEV]
nrf802154_sniffer.py: error: unrecognized arguments: --extcap-version
10:51:27.892  Capture Warn sync_pipe_wait_for_child: waitpid returned EINTR. retrying.

The last 'warning' repeats several times. I am filtering using zbee-zcl and only see broadcast packets even though there is obviously stuff going on within the same channel as i have specified.

As soon as I try to stop capturing Wireshark crashes.

Any ideas anyone? I am new to Wireshark and Zigbee trying to get my head around it all.

e-rk, 2019-01-24:
Hi,
It appears that Wireshark 2.9 sends its version as an additional command line parameter to let the extcap utility adapt its behaviour to different Wireshark versions, which is a bit unfortunate in this case.
I'll let you know once a fix is up.

IanAber, 2019-01-25:
> Now that I think of it the unrecognized argument is pretty harmless in this case.
> Unfortunately I have no idea why Wireshark crashes after capture on Mac and I have no way to verify it myself. If you run Wireshark from the terminal do you see any error logs when trying to stop the capture? I think this problem should be reported to Wireshark developers.
>
> About the ZigBee broadcasts, do you have correct decryption keys set up? The sniffer does no data processing of sniffed packets so I suspect that it is a matter of changing the correct preferences in Wireshark. I am not familiar with ZigBee, so I can't help here either. Sorry.

I found that running Wireshark as an administrator (using sudo wireshark from a terminal window) fixes the EINTR errors so it must be a permissions thing. My problem now is that as soon as I stop capture Wireshark crashes with a segmentation fault. I have posted a question on the Wireshark site and the admin asked where I got the Python script. I sent him the link to the Github site for NordicPlayground. I am fairly sure he is going to tell me it is a problem with the external capture piece so I imagine I am going to have to debug this bit myself.

Is it possible to get the sniffer to output directly into a terminal window or something outside of Wireshark so I can test out the capture and try feeding the resulting data info Wireshark as a file? I connected a terminal to the USB serial port and it looks like it should respond to some AT type commands. Do you know what I can send to get it to dump stuff out, if that is possible?
Ian Abercrombie

Ian.Abercrombie@CedarTechnology.com <mailto:Ian.Abercrombie@CedarTechnology.com>
Phone: +1 (352) 281-6286
www.CedarTechnology.com <http://www.cedartechnology.com/>

e-rk, 2019-01-25:
If you are in the repository root, you can install the module using the command:
```
python -m easy_install .
```
You might need root privileges for that.
Then the `nrf802154_sniffer` module can be imported into other python scripts. It exposes the `Nrf802154Sniffer` class. All you need to do is to call `extcap_capture` and give it the file name, serial port and channel as arguments. To stop the capture call `stop_sig_handler()`.

stig-bjorlykke, 2019-01-25:
The crash on stop capture is [Wireshark bug 14631](https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=14631).

IanAber, 2019-01-28:
I managed to get past the Wireshark crash issue by downloading the source and recoiling Wireshark from scratch. Like everything els I have done recently it was not a smooth operation as the instructions didn’t cover all the requirements and the build scripts failed. I found the missing libraries and installed them and now Wireshark is recording packets correctly without blowing up when I stop the capture.

The next problem on the ridiculously steep learning curve to figuring this out, is how to get Wireshark to capture anything other than broadcast packets. What do I need to tell Wireshark to get all Zigbee packets to show up?

I have a Samsung Smartthings hub which s connected to a half a dozen Zigbee devices. One of these devices is a power outlet and I can turn it on and off via my iPhone app. This must be sending Zigbee packets to the socket but they do not show up in Wireshark. Neither do any of the network pulse checking packets that are being sent to confirm availability. All I see is Broadcast packets…

"34","2400.930382","","Broadcast","IEEE 802.15.4","8","Beacon Request”

What do I need to be able to see all communication from hub to device and back?

I really appreciate your help and feel guilty that I have been unable to work it out myself.
Ian Abercrombie

Ian.Abercrombie@CedarTechnology.com <mailto:Ian.Abercrombie@CedarTechnology.com>
Phone: +1 (352) 281-6286
www.CedarTechnology.com <http://www.cedartechnology.com/>

IanAber, 2019-07-29:
Version 3.0.3 gets past the Wireshark crash issue.

closed 2019-07-29 by IanAber

stig-bjorlykke, 2019-07-30:
@IanAber Good. I have closed [Wireshark bug 14631 ](https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=14631).