nordicsemi/nRF-Sniffer-for-802.15.4 #21: USB device not recognised

cmcclenaghan, 2019-03-01:
The VID and PID are not recognised from either the DK or the dongle devices.
These appear as VID_1915 and PID_154A but I can't find which USB driver provides these so the the sniffer is not listed by `serial.tools.list_ports.comports()`

Can you advise on the driver to use?

e-rk, 2019-03-01:
Hi,
The operating system should use the USB CDC ACM driver to communicate with the device. On Windows 10 and most Linux distros these drivers should be available "out of the box".
What operating system do you use?

cmcclenaghan, 2019-03-01:
Hi. It's with Window 7 64-bit.

e-rk, 2019-03-01:
Windows 7 requires a digitally signed .inf file to be installed to associate the device with the USB serial driver. Unfortunately we do not have such file available right now but we are going to deliver it.
I will let you know when we have it ready.

cmcclenaghan, 2019-03-01:
Thanks for the update and your prompt responses.
Have you an estimation of when you might deliver that?

HNOTT, 2019-03-06:
@cmcclenaghan There is a .inf file hanging about in the nRF SDK that worked for me. I found it in nRF5 SDK for Thread and Zigbee v2.0.0\examples\usb_drivers. I had to **force** Windows (I'm using Windows 7) to use it to update the driver of the nRF52840 dongle with the Zigbee sniffer firmware on it. I choose the "nRF52 USB CDC Demo".

@e-rk I tried using the nRF52840 dongle with the Zigbee sniffer firmware on it with Windows 10 and it did exactly the same as with Windows 7. It could not find the driver. Did not try installing the .inf from the SDK though.

cmcclenaghan, 2019-03-06:
@HNOTT Thanks, I thought I had used that location for the other applications but I'll go back and hunt again.

e-rk, 2019-03-07:
@HNOTT Hi. If you see a single unrecognized device in the device manager with the dongle connected then most likely it is the DFU trigger interface.
The sniffer firmware for the dongle should create two interfaces on Windows 10, one for triggering DFU and the second for sending captured packets. The capture should work fine even without the DFU trigger installed.

Please verify if there are any new COM ports appearing when you plug the device.

To fix the unrecognized device warning, install the Nordic DFU trigger driver. It should be available after installing nRF Connect for desktop. Click the Update driver on the interface with the warning sign in the device manager and manually select the Nordic DFU trigger driver from the list.

@cmcclenaghan As for the estimation we do not have a precise date right now. Hopefully we will be able to sign the file this month, but I can't promise. The solution mentioned by @HNOTT should work but it may require slight modifications to the `inf` file and configuring the Windows to allow unsigned drivers.

e-rk, 2019-08-06:
The USB driver is provided in the nRF5 SDK for Thread and Zigbee under `examples/usb_drivers` directory.

closed 2019-08-06 by e-rk