nordicsemi/nRF-Sniffer-for-802.15.4 #27: Remove DFU Trigger Library from the firmware

LuDuda, 2019-03-08:
Due to different problems (bad user experience) with autodetection of correct drivers of DFU trigger in Windows, I suggest to remove it. Flashing of nRF52840 Dongle is anyway usually one-time operation, after which you need to re-plug dongle manually. Thoughts?

hubertmis, 2019-03-08:
I don't think I can answer this question. I'm not an expert with DFU trigger or uwing nRF52840 Dongle in Windows. I think I flashed it once or twice in Linux using reset button.

e-rk, 2019-03-08:
Personally I have used the DFU trigger a couple of times during firmware development, but I have the dongle always nearby, so for me it is not a problem press reset. I have to replug the device manually anyway.

LuDuda, 2019-03-08:
If there is no objection.. @e-rk can you please provide PR to fix this issue, before the release?

e-rk, 2019-03-08:
@LuDuda I can commit the new firmware to the #22 branch.

e-rk, 2019-03-18:
Resolved by #22.

closed 2019-03-18 by e-rk