nordicsemi/nRF-Sniffer-for-802.15.4 #35: Channel always goes to 11 after a reset in Wireshark with no way to change.

IanAber, 2019-07-30:
When you start the sniffer you can click the gear icon to the left of the source selector and set a channel number. This will remember the channel selected from one run of Wireshark to another but if you simply reset the trace using the toolbar icon 'Restart Current Capture' the channel will change to channel 11 and there does not appear to be a way to change it back without exiting the entire program and starting again.

e-rk, 2019-07-30:
To me it seems that there might be some problem with writing the channel setting to Wireshark's preferences file.
Can you please run `tshark -G currentprefs` and verify the value of `extcap._dev_ttyacm0.channel` property before and after starting the capture?
On my system the value changes from the default channel 11 to the desired channel after I start the capture.
If the value doesn't change, there might be some problem with write permissions to the preferences file.

IanAber, 2019-07-31:
I checked the settings using the shark -G currentprefs command as you describe and I see that the value for the channel is correctly recorded as 14.

If I start Wireshark and begin a capture session by selecting the Nordic dongle without changing any of my settings it begins capturing on channel 14 as I would expect. If I then click the light green restart current capture button third from the left on the toolbar, the dongle switches to channel 11, which is the default, and now captures traffic from my other Zigbee network. Closing out of Wireshark and starting again it will initially capture on channel 14 as defined in my preferences.

If I edit the Python script to change the default to channel 14 and I set my preferences to 11 on starting afresh the dongle will capture on channel 11 but on clicking restart will switch to channel 14. This suggests to me that the restart is not using the value found in any preferences file but simply restarting the capture using the hard coded value in the Python script.

e-rk, 2019-08-06:
Hi. Sorry for taking this long to respond.
Can you please check the command line arguments that the extcap script is started with by the Wireshark? The argument of interest would be `--channel`. If the channel after resetting the capture is a correct one and the sniffer still listens on channel `11` for some reason it would definitely be an extcap issue that we would have to look into. So far I have no clue what is going wrong.