nordicsemi/nRF-Sniffer-for-802.15.4 #37: [question] 802.15.4 capabilities

ikarus23, 2019-11-09:
Hi, thanks for this helpful project!

Maybe this is a stupid question and it is mentioned somewhere in the datasheet of the nRF dongle, but what parts of the 802.15.4 standard are supported? Is it a specific version? Like 2011? What baud rates? What transmission methods (DSSS vs CSS)?  Using nRF52840 I guess it's 2.4GHz only, or is it?

Would be great to have some additional information in the readme. ;)

e-rk, 2019-11-12:
Hi,
The sniffer captures any IEEE 802.15.4 frame as long as it is transmitted using O-QPSK 2.4GHz PHY with DSSS. The specified transmission rate for that PHY is 250kb/s.
The sniffer should not care about any specific version of the standard as frame dissection is done on the PC side.

You are right about including this information in the readme, thank you for the suggestion.

ikarus23, 2019-11-12:
Hi @e-rk,
thanks for clearing that up. I will close this issue.

Just out of curiosity: Do you know of a cheap, flexible and ready to use platform to sniff 802.15.4 with CSS?

closed 2019-11-12 by ikarus23

e-rk, 2019-11-13:
I'm not aware of any such platform, sorry.