nordicsemi/nRF-Sniffer-for-802.15.4 #41: Q: Is the source code for the hex files available?

markus-becker-tridonic-com, 2020-01-29:
Is the source from which the hex files available? Is the code based on Connect SDK? Zephyr?

hubertmis, 2020-01-29:
The source code is not publicly available. It is a simple application that uses:
* nRF 802.15.4 radio driver: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver
* USB CDC-ACM driver from nRF5 SDK

markus-becker-tridonic-com, 2020-01-29:
Hmm, OK. Would be nice to have a sample in Zephyr that would work with the Wireshark plugin.

hubertmis, 2020-01-29:
That's a good idea, we'll keep it in mind during integration of other applications with Connect SDK.

swattor, 2020-02-13:
It would be great if this could be considered for opensource!

markus-becker-tridonic-com, 2020-02-17:
Possibly sniffing could also be done using https://github.com/openthread/pyspinel/blob/master/sniffer.py based on an OpenThread firmware. Not sure how this compares feature-wise.

Leo2442926161, 2020-09-25:
hope it will be open source.

cederom, 2020-11-29:
+1 for full open-source :-) I am switching from ARM MBED to ZEPHYR after your online webinars showing how ZEPHYR works and I must admit this is something I was waiting for on my FreeBSD workstation! =)

dberliner, 2021-09-13:
Is there anything in the way of this being open source?

hecko, 2022-06-17:
+1 to see sources here. especially if sources are based on Zephyr

bm16ton, 2022-11-18:
+1 on opensource, realisticly i caint imagine any end user wouldnt prefer that. Consider every unit sold to end users a +1. Ive never been able to get a good answer from companies on reasons for not opensource small basic things like this. What could it hurt? How could you loose money? Possible this is a hold over thought process from back when Balhmer was saying things like people who use open source are terrorists kinda days? In reality more sniffer projects around more people see them and more people buy the product, plus more people look at the support libs etc and PR fixes/improvements, thats free laybor. Plus the thing that realy confuses me, when a product/firmware etc like this is released and kept closed source/no help for people who are writing say open source drivers etc. The company knows the driver will get written, the guy writting the driver knows this as well,the only difference it takes sum poor programmar Bob more time to achieve his goal..So the only reason not to help out is because you want to waiste poor Bobs time (and all the people in this issues thread plus counteless others) Thats it. We can reproduce this firmware you know we can, so any concerns you have are gonna happen once we do,the only difference is you prefer to waiste the time of poor opensource programmers who bought your products. Poor poor Bob..

ckielstra, 2024-03-06:
May 2023, a compatible [sniffer code](https://github.com/nrfconnect/sdk-nrf/tree/main/samples/peripheral/802154_sniffer) example was added to the Nordic [sdk-nrf](https://github.com/nrfconnect/sdk-nrf) project.

dberliner, 2024-03-21:
FYI at the time of this writing the sample code does not support Wireshark.

e-rk, 2024-04-05:
The source code of the firmware is available in the nRF Connect SDK. See readme in this repository.

markus-becker-tridonic-com, 2024-04-08:
Awesome. Thanks @e-rk !