nordicsemi/nRF-Sniffer-for-802.15.4 #43: nRF52840 DONGLE DFU ZIP

cederom, 2020-11-29:
Hello world :-)

Can you please provide instructions on how to convert HEX into DFU ZIP so I can use `nrfutil` python utility?

Can you please provide a DFU ZIP package along the HEX file for a DONGLE just to quickly flash it using `nrfutil`?

I am using FreeBSD and only Open-Source stuff works here :-)

Thank you :-)
Tomek

cederom, 2020-11-29:
https://github.com/NordicSemiconductor/pc-nrfutil/issues/335

cederom, 2020-11-29:
It seems I made the debug package with a blind guess of parameters just to make things work.. but it would be nice to have this nice DFU ZIP package provided too for platforms where only `nrfutil` is available :-)

```
nrfutil  pkg generate --debug-mode --application sniffer_nrf52840dongle_nrf52840_7cc811f.hex --hw-version 52 --sd-req 0 sniffer_nrf52840dongle_nrf52840_7cc811f.zip
Failed to import ecdsa, cannot do signing

|===============================================================|
|##      ##    ###    ########  ##    ## #### ##    ##  ######  |
|##  ##  ##   ## ##   ##     ## ###   ##  ##  ###   ## ##    ## |
|##  ##  ##  ##   ##  ##     ## ####  ##  ##  ####  ## ##       |
|##  ##  ## ##     ## ########  ## ## ##  ##  ## ## ## ##   ####|
|##  ##  ## ######### ##   ##   ##  ####  ##  ##  #### ##    ## |
|##  ##  ## ##     ## ##    ##  ##   ###  ##  ##   ### ##    ## |
| ###  ###  ##     ## ##     ## ##    ## #### ##    ##  ######  |
|===============================================================|
|You are generating a package with the debug bit enabled in the |
|init packet. This is only compatible with a debug bootloader   |
|and is not suitable for production.                            |
|===============================================================|

|===============================================================|
|##      ##    ###    ########  ##    ## #### ##    ##  ######  |
|##  ##  ##   ## ##   ##     ## ###   ##  ##  ###   ## ##    ## |
|##  ##  ##  ##   ##  ##     ## ####  ##  ##  ####  ## ##       |
|##  ##  ## ##     ## ########  ## ## ##  ##  ## ## ## ##   ####|
|##  ##  ## ######### ##   ##   ##  ####  ##  ##  #### ##    ## |
|##  ##  ## ##     ## ##    ##  ##   ###  ##  ##   ### ##    ## |
| ###  ###  ##     ## ##     ## ##    ## #### ##    ##  ######  |
|===============================================================|
|You are not providing a signature key, which means the DFU     |
|files will not be signed, and are vulnerable to tampering.     |
|This is only compatible with a signature-less bootloader and is|
|not suitable for production environments.                      |
|===============================================================|

Zip created at sniffer_nrf52840dongle_nrf52840_7cc811f.zip
```

Flashed it with 

```
 nrfutil --verbose dfu usb-serial -p /dev/cuaU0 -pkg sniffer_nrf52840dongle_nrf52840_7cc811f.zip
Failed to import ecdsa, cannot do signing
  [####################################]  100%
Device programmed.
```

As a result I got umodem device up and running :-)

```
ugen0.12: <ZEPHYR nRF Sniffer for Bluetooth LE> at usbus0
umodem0 on uhub0
umodem0: <ZEPHYR nRF Sniffer for Bluetooth LE, class 0/0, rev 2.00/2.03, addr 30> on usbus0
umodem0: data interface 1, has CM over data, has no break
```

The OS is FreeBSD :-)

```
FreeBSD 0xCFMX4 12.2-RELEASE FreeBSD 12.2-RELEASE r366954 GENERIC  amd64
```

hubertmis, 2020-11-29:
Thank you for reporting this feature request.

The Device Firmware Upgrade (DFU) process supported by `nrfutil` is a feature for deployed products where firmware can be automatically or semi-automatically updated by end-users in a secure way. Most devices connected to internet (like smartphones, home automation devices, etc) sometimes ask to update their firmware. DFU from nrfutil is a feature that helps to implement secure firmware update procedures in end products. It involves security operations to ensure that the firmware being updated was created by the device vendor, to prevent firmware version rollback, etc.

nRF Sniffer for 802.15.4 is intended to be a simple development tool. It is not intended to be deployed as a part of a final products. Because of that DFU or FOTA are overkill for this tool and these features are not planned in its roadmap.

The idea is that an engineer that intends to use the sniffer, uses nrfjprog (from nRF Command Line Tools package) or nRF Connect for Desktop to flash a sniffer device. If an engineer is interested in flashing newer firmware version, he/she should use nrfjprog or nRF Connect for Desktop again. However, the firmware is released rarely and because of that flashing newer firmware version is not an operation performed often by users.

hubertmis, 2020-11-29:
@cederom, I've just read in https://github.com/NordicSemiconductor/pc-nrfutil/issues/335 that you are interested in Bluetooth protocol. Please not that nRF Sniffer for 802.15.4 is a sniffer of 802.15.4 packets. 802.15.4 and Bluetooth are incompatible radio protocols.

Have a look at this tool to sniff Bluetooth protocol: https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE

> As a result I got umodem device up and running :-)
> 
> ```
> ugen0.12: <ZEPHYR nRF Sniffer for Bluetooth LE> at usbus0
> umodem0 on uhub0
> umodem0: <ZEPHYR nRF Sniffer for Bluetooth LE, class 0/0, rev 2.00/2.03, addr 30> on usbus0
> umodem0: data interface 1, has CM over data, has no break
> ```

It seems that you have nRF Sniffer for Bluetooth LE running. It is other tool than nRF Sniffer for 802.15.4. I believe you are interested in nRF Sniffer for Bluetooth LE, not 802.15.4.

However, if you need any help with nRF Sniffer for 802.15.4, please add your questions in this repository, or reopen this issue.

closed 2020-11-29 by hubertmis

cederom, 2020-11-29:
Hello @hubertmis and thank you for quick response! Whoops, so this 802.15.4 sniffer is something different than BT sniffer, sorry and thanks for pointing this out, my first steps here :-)

If possible I am still interested in a DFU ZIP for DONGLE for this 802.15.4 (also for the BT project) :-)

There is only `nrf802154_sniffer_dongle.hex` provided and this cannot be uploaded to the DONGLE without external debug probe while there is a still DFU mode that we could use, right?

nRF Connect is not available on my platform so I need to use `nrfutil` and I stick only to the Open-Source :-)

cederom, 2020-11-29:
Well I can talk to the 802.15.4 sniffer over minicom terminal! I have generated DFU ZIP exactly the same way as described above. This probably means a problem with BT sniffer firmware. Maybe `--sd-req` needs something else than `0`.

Why there is only one dedicated repo on github for 802.15.4 while not for BT? These solutions are very similar from the computer perspective :-)

Thank you for your feedback and time @hubertmis :-)

hubertmis, 2020-11-30:
> If possible I am still interested in a DFU ZIP for DONGLE for this 802.15.4 (also for the BT project)

I've explained above why DFU is overkill for the sniffer project. We don't plan to add such feature. 

> this cannot be uploaded to the DONGLE without external debug probe

You can upload firmware to dongle using nRF Connect for Desktop. You don't need debug probe or nrfutil.

> nRF Connect is not available on my platform so I need to use `nrfutil`

Perhaps that's the source of the problem. I recommend to create a devzone ticket to run nRF Connect for Desktop on your platform.

> Maybe `--sd-req` needs something else than `0`.

I can't help you with bluetooth sniffer. Here again I recommend to create a devzone ticket. 

> Why there is only one dedicated repo on github for 802.15.4 while not for BT?

The goal of these tools is similar, but the implementation is different. We are in the process of unification distribution methods of all tools delivered by Nordic. Currently some tools are deployed to GitHub, while other are not.