nordicsemi/nRF-Sniffer-for-802.15.4 #44: FreeBSD umodem detect baudrate

cederom, 2020-11-29:
Hello world :-)

I have problem running Wireshark Sniffer using nRF52840 DONGLE on FreeBSD. It seems that umodem kernel driver only accepts given set of baudrates here. This results in `UART.find_sniffer(write_data=False)` returning empty port list and so no device is visible in WireShark.

I have managed to flash HEX into the DONGLE using `nrftool` and some blind DFU ZIP conversion. Can this be the cause of problem?

Is there any way I can talk to the DONGLE Sniffer over MiniCom (serial terminal application) in order to verify its operations? 

Is there any python script that could veryfy operations of a Sniffer on a given port (`/dev/cuaU0` in my case)?

Any hints welcome :-)
Tomek

cederom, 2020-11-29:
Is it possible to set baudrate on the Sniffer side to one of `[230400, 403200, 460800, 806400, 921600, 1500000, 3000000]` just to verify if the problem is related with baudrate rejection by kernel driver?

cederom, 2020-11-29:
Can you please specify exact port parameters on the Sniffer firmware side?

hubertmis, 2020-11-29:
Thank you for sharing your experience with running 802.15.4 Sniffer on FreeBSD.

> I have managed to flash HEX into the DONGLE using `nrftool` and some blind DFU ZIP conversion. Can this be the cause of problem?

I have no experience with flashing the dongle using `nrftool`. I've flashed mine using nRF Connect for Desktop. As long as the hex file is correctly stored in the dongle memory, I think the tool used to flash makes no difference. However, to make sure, I would recommend to try flashing the dongle using a PC running one of operating systems supported by nRF Connect for Desktop and use it with Wireshark running in FreeBSD. When you manage to run this combination, you could try flashing in FreeBSD itself and verify if everything works as expected.

> Is there any way I can talk to the DONGLE Sniffer over MiniCom (serial terminal application) in order to verify its operations?

Yes. You can use a serial terminal like minicom, screen, putty, etc and connect to the sniffer device. It exposes a text command line interface. You can try typing a command `channel` followed by a newline character (enter key). The CLI interface is supposed to echo each character you type and report the default channel (`11`) after you use this command.

> Is there any python script that could veryfy operations of a Sniffer on a given port (`/dev/cuaU0` in my case)?

The python module compatible with the sniffer command line interface is included in this repository: https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4/blob/master/nrf802154_sniffer/nrf802154_sniffer.py . It can be used like Wireshark extcap process (command line options compatible with extcap interface), or imported in other python programs.

> Can you please specify exact port parameters on the Sniffer firmware side?

The sniffer uses USB CDC as a serial transport. It does not use UART. Because of that it is not needed to select baud rate or any other UART parameter when connecting to the device.

cederom, 2020-11-29:
Allright I will find another machine to run nRF Connect to compare the results and will report back but ultimately the `nrftool` is the only supported utility on my platform and I would like to keep everything in one place I am sure this is possible to achieve, thanks for the hints @hubertmis  :-)

cederom, 2020-11-30:
* Using the same setup I was able to convert 802.15.4 Sniffer HEX to DFU ZIP and flash it onto the nRF52840 DONGLE.
* I can talk to 802.15.4 Snifferover minicom serial terminal no matter what port speed. I can see a nice menu unlike the BLE Sniffer which does not respond at all. 
* True it is not a hardware UART speed does not matter.
* I can confirm with 802.15.4 firmware, this seems to be a problem with BLE Sniffer firmware.

Technically speaking this case is solved for 802.15.4 Sniffer so I will close the ticket. Thank you @hubertmis  :-)

Now I just need to get Wireshark excap find that 802.15.4 Sniffer :-)

closed 2020-11-30 by cederom