nordicsemi/nRF-Sniffer-for-802.15.4 #45: Startup time is too long when start capturing packets

xiaolongba, 2021-01-07:
Hi experts,

I followed the instructions the Readme file described.  It almost works well. But I still found two issues:

1. the Startup time is too long when pressing the <code>Start capturing packets</code> button, it took me around 15 seconds. my environment setups as below:
  - python 3.9.1
  - Wireshark 3.4.2
  - using the firmware here
 
For the details pls refer to the following plot:

[image]

can you give me some tips to optimize it?

2. Wireshark will exit once I stop capturing packets when there are BLE and 802.15.4 in the <code>extcap folder </code>together even if there is only 802.15.4 sniffer connects to the PC. let me show you more details as below:
[image]

[image]

can you give me a hand?

thanks in advance

hubertmis, 2021-01-07:
Thanks for reporting this issue.
I recommend posting this problem to [Nordic DevZone](https://devzone.nordicsemi.com/). It would be a faster way to get assistance.

When you create a DevZone ticket, please add a link to the ticket here for reference. It would help others, if anyone else would hit the same issue.

xiaolongba, 2021-01-07:
Hi @hubertmis 

thanks for your suggestions, I reported the same issue on Nordic DevZone already. this ticket is [here](https://devzone.nordicsemi.com/f/nordic-q-a/70211/startup-time-is-too-long-when-using-802-15-4-sniffer-to-start-capturing-packets)

atsju, 2025-01-10:
There are 2 issues in this issue. 
1. slow/long startup
2. coexistence bug between BLE and 802.15.4 sniffer

I have the first issue too. Reported to Nordic DevZone [here](https://devzone.nordicsemi.com/f/nordic-q-a/117937/nrf-sniffer-for-802-15-4-slow-startup) let's get this fixed, it's 4 years old.

Notifying @e-rk @greg-fer @stig-bjorlykke @pszkotak @hubertmis as they have all contributed to this tool.

e-rk, 2025-01-20:
Fix available here: https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4/pull/75

PR #75 (2025-01-21): extcap: Refactor the extcap script
Improved the implementation of the extcap script.
* The script now has significantly faster startup time.
* Removed the old LUA dissector and the associated DLT.
* IEEE 802.15.4 TAP is now the default capture mode.

Requires Wireshark 4.0 or later.

Fixes #45 
Closes #74
closed 2025-01-21 by e-rk