nordicsemi/nRF-Sniffer-for-802.15.4 #56: nRF52840 support for tshark?

Geobm, 2023-02-13:
First of all, thanks for the great project.

Currently I'm doing some test with a nRF52840 dongle (as sniffer) and a NRF52840-DK as as a peripheral and using tshark with python to analyze the packets from a terminal output (to automate the sniffing instead of using wireshark GUI). Basically I just want to set the first `CONNECT_IND` frame and once I receive a `LL_VERSION_IND`, `LL_FEATURE_REQ` or `LL_LENGTH_REQ`, extract the time but in my python script.  As far I been trying it seems that tshark misses some packets that I try to send (such as CONNECT_IND). In fact I made some modifications to the nrf802154_sniffer.py script that only will listen to my desired advertising channel (39) and forced to only sniff my testing address (3c:61:05:4c:33:6).  I modified this function as follows:

```python3
def device_added(notification):
    global sniffer
    """A device is added or updated"""
    device = notification.msg
    #logging.info(device)
    # Only add devices matching RSSI filter
    if rssi_filter == 0 or device.RSSI > rssi_filter:
        # Extcap selector uses \0 character to separate value and display value,
        # therefore the display value cannot contain the \0 character as this
        # would lead to truncation of the display value.
        display = (device.name.replace('\0', '\\0') +
                   ("  " + str(device.RSSI) + " dBm  " if device.RSSI != 0 else "  ") +
                   string_address(device.address))

        message = str(device.address) + '\0' + display
        if list(device.address) == [60, 97, 5, 76, 51, 110, 0]:
            logging.info("GOT MATCH ----------------------------")
            follow_address(sniffer, str('3c:61:05:4c:33:6e public'))
        logging.info(device.address)
        control_write(CTRL_ARG_DEVICE, CTRL_CMD_ADD, message)
```

Basically this line in my script opens tshark  : `tshark_output = subprocess.Popen(['tshark', '-i', '12', '-f', 'stdout=subprocess.PIPE)`, that listen to my interface 12. When sniffing with wireshark the packets seems to have a normal behavior. Nevertheless, in tshark I can only see the advertisements and scan requests. 

Finally mi question is, does nRF52840 dongle and nRF52840 DK  have support for tshark? if not, I would really appreciate any idea to approach this. TIA

e-rk, 2023-02-13:
I'm a bit confused here.
If I understand correctly, you are trying to sniff a specific **Bluetooth** connection. Meanwhile the nrf802154_sniffer can capture only the **IEEE 802.15.4** traffic and is unable to capture Bluetooth frames.
Did you mean to receive support with the [BLE sniffer](https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le) instead? If yes, then you should make a ticket on the [devzone](https://devzone.nordicsemi.com/) as here we can only help with the nrf802154_sniffer.

Geobm, 2023-02-13:
Completely my mistake @e-rk , I thought it was the nRF Sniffer repository for Bluetooth. Thank you very much, closing this issue.

closed 2023-02-13 by Geobm