nordicsemi/nRF-Sniffer-for-802.15.4 #62: Ubuntu 22.04 LTS nrf802154_sniffer.py problem. nRF Sniffer isn't listed as an interface

napent, 2024-04-28:
As Ubuntu 22.04 LTS is lacking python alias the 
```#!/usr/bin/env python```
line should be replaced with
```#!/usr/bin/env python3```
Another resoulution would be to add `sudo apt install python-is-python3` as a requirement.

e-rk, 2024-04-29:
I think the preferred solution is to install `python-is-python3` on Debian-based distros. Other Linux distributions might not have `python3` alias on the other hand.

Chapoly1305, 2024-04-30:
No idea somehow this issue occurred from nowhere, I had it running fine and it just stopped working. Anyway, I don't think anyone is still using older version, it shall be modified. https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4/pull/63

WayneKeenan, 2024-05-05:
To avoid system wide changes (now and any unexpected ones in the future) I created a python3 venv in my home/project folder to install pyserial and hardcoded the plugins shebang to point the venv python3 bin.

e-rk, 2024-05-06:
@Chapoly1305 regarding #63 I'm a bit worried about changing the executable name, as it might break other systems such as macOS. I think the solution that should work across all systems would be to create an [entry point](https://setuptools.pypa.io/en/latest/userguide/entry_point.html) and install the sniffer using `pip` or `pipx` (the latter automatically creates virtualenv for each application). `pip` or `pipx` would ensure that the correct interpreter executable is used and all dependencies are installed.
The sniffer command would then be accessible from `PATH` and the user could copy `nrf802154_sniffer.sh` to Wireshark extcap directory.
The `nrf802154_sniffer.sh` file doesn't exist right now, but it would be similar to [this one](https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4/blob/master/nrf802154_sniffer/nrf802154_sniffer.bat).

Chapoly1305, 2024-05-06:
@e-rk  Good Point! I have checked on my macbook and windows, it seems the python3 can be recognized well. python is not recognized neither on macbook. 

[image]

[image]

e-rk, 2024-05-07:
Fixed in https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4/pull/63

closed 2024-05-07 by e-rk