nordicsemi/nRF-Sniffer-for-802.15.4 #68: Wireshark on MacOS not seeing capture device

edpgcooper, 2024-06-21:
My dongles work on windows, however on Mac OS (Monterey 12.7.5) with Python 3.11.9 Im not having much luck.
Installed the nrf802154_sniffer.py in the extcap folder, when I run 
`./nrf802154_sniffer.py --extcap-interfaces`
I get
`extcap {version=0.7.2}{help=https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4}{display=nRF Sniffer for 802.15.4}
interface {value=/dev/cu.usbmodem14101}{display=nRF Sniffer for 802.15.4}
interface {value=/dev/cu.usbmodem14103}{display=nRF Sniffer for 802.15.4}
control {number=6}{type=button}{role=logger}{display=Log}{tooltip=Show capture log}`

Any ideas why I have the double entry, don't seem to get this on Windows?

Chapoly1305, 2024-06-21:
sudo is required. Have you tried `sudo ./nrf802154_sniffer.py --extcap-interfaces`

edpgcooper, 2024-06-21:
> sudo is required. Have you tried `sudo wireshark`

Thanks, 'sudo open /Applications/Wireshark.app ' works for me, must have missed in the documentation where I have to do that, bit confused why also!  Thanks tho

stig-bjorlykke, 2024-07-03:
Running Wireshark with `sudo` is never a good idea.

edpgcooper, 2024-07-03:
What am I meant to do instead then?  The interface only appears if I do.

stig-bjorlykke, 2024-08-05:
> What am I meant to do instead then? The interface only appears if I do.

You should have a look at the device permissions. Mine looks like this.
```
crw-rw-rw-  1 root  wheel  0x900000f Aug  5 10:30 /dev/cu.usbmodem8401
```

motters, 2024-09-13:
This fixed it for me, `/usr/bin/pip3 install -U pyserial` 

Credit to the Nordic DevForum somewhere.

SLboat, 2025-02-03:
I just ran into this problem, and what caused it was homebre's python3 overriding the system default, which had the consequence that the system's own python3 was missing the library.

It's also possible to modify the environment in which py scripts run like this:
[image]

```
~ ➤ which -a python3                                                        
/opt/homebrew/bin/python3
/usr/bin/python3
~ ➤ /usr/bin/python3 -m pip list                                            
Package           Version
----------------- -------
altgraph          0.17.2
astroid           2.13.2
dill              0.3.6
future            0.18.2
isort             5.11.4
lazy-object-proxy 1.9.0
macholib          1.15.2
mccabe            0.7.0
ping3             4.0.4
pip               21.2.4
platformdirs      2.6.2
pylint            2.15.10
setuptools        58.0.4
six               1.15.0
tomli             2.0.1
tomlkit           0.11.6
typing_extensions 4.4.0
wheel             0.37.0
wrapt             1.14.1
~ ➤ /usr/bin/python3 -m serial                                                        
/Library/Developer/CommandLineTools/usr/bin/python3: No module named serial

~ ➤ python3 -m serial                                                                 

--- Available ports:
---  1: /dev/cu.Bluetooth-Incoming-Port 'n/a'
---  2: /dev/cu.debug-console 'n/a'
--- Enter port index or full name: q
could not open port 'q': [Errno 2] could not open port q: [Errno 2] No such file or directory: 'q'
```