nordicsemi/nRF-Sniffer-for-802.15.4 #70: Most recent 'nrfutil device' programmer can't read included .hex firmware files anymore / This repo unsupported?

EskoDijk, 2024-09-28:
Firmware files are included in `.hex` format in this repo. The most recent programmer 'nrfutil device' can't read this format, it seems.

Programmer version:

```
~/sniffer$ nrfutil device -V
nrfutil-device 2.5.1 (3e5d1ee 2024-09-03)
commit-hash: 3e5d1eec241ea7c30250d2e082cbaf111f6b3dea
commit-date: 2024-09-03
host: x86_64-unknown-linux-gnu
build-timestamp: 2024-09-03T13:46:41.857868289Z
classification: nrf-external
```

Error message:

```
~/sniffer$ nrfutil device program --firmware nrf802154_sniffer/nrf802154_sniffer_nrf52840dongle.hex 
[00:00:00] ------   0% [D02B62C6E79F] Failed, [sdfu] firmware file extension .hex is not supported
Error: One or more program tasks failed:
 * D02B62C6E79F: [sdfu] firmware file extension .hex is not supported, code: Argument
```

The programmer does work for .zip firmware files though but these aren't included in this repo.

Answer from Nordic support is to use another source of the firmware: https://devzone.nordicsemi.com/f/nordic-q-a/114969/nrfutil-device-program-results-in-sdfu-firmware-file-extension-hex-is-not-supported 

From which it sounds like this repo should be deprecated?

e-rk, 2024-09-30:
I suggest using the [nRF Connect for Desktop](https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-Desktop) to program the dongle.

>Answer from Nordic support is to use another source of the firmware

The support ticket mentions nRF Sniffer for Bluetooth LE.
This repository is for the nRF Sniffer for IEEE 802.15.4,  which is a different radio protocol.

EskoDijk, 2024-10-01:
> The support ticket mentions nRF Sniffer for Bluetooth LE. This repository is for the nRF Sniffer for IEEE 802.15.4, which is a different radio protocol.

Thanks! Yes, although it was for another protocol the issue hit was quite similar. It can be solved also by using a .zip version of the same hex file - so in principle also the .zip file versions could be added to this repository as an alternative to installing the desktop app. (I have this working now based on a .zip I got via a Thread Group Nordic contact.)

Damian-Nordic, 2025-02-10:
There's another `nrfutil` command that can be used to convert HEX files to ZIP (after running `nrfutil install nrf5sdk-tools`):
```
nrfutil nrf5sdk-tools pkg generate --hw-version 52 --sd-req=0x00 \
    --application "fw.hex" \
    --application-version 1 "fw.zip"
```